fablabbcn/smartcitizen-web #273: Tags filter doesn't work properly on the new version

pral2a, 2018-03-14:
## Issues on staging

- [ ] Screen doesn't show on the proper location and causes a weird problem on scroll

[image]

- [ ] Tags labels are too small, check the original

[image]

- [ ] **Tags doesn't show up**

[image]

pral2a, 2018-03-19:
- [ ]  Tags doesn't show up 

Remains open

viktorsmari, 2018-03-19:
Works when selecting 2 tags, but not 1 tag.
Also, check when removing only 1 tag

viktorsmari, 2018-03-20:
We get this error when we use only 1 tag:
`TypeError: Cannot assign to read only property 'tags' of object '#<Object>'`

`$stateParams.tags = [$stateParams.tags];`

https://github.com/fablabbcn/smartcitizen-web/blob/master/src/app/app.route.js#L144

PR #277 (2018-04-05): Tags: check if multiple tags or a single one
Closes #273
closed 2018-04-05 by viktorsmari