fablabbcn/smartcitizen-web #290: Gulp build is not working after commit fc435b2

viktorsmari, 2018-05-16:
### Expected behaviour
`gulp build` only works in one of those commands:

`gco .; gco fc435b2; npm i; bower i; gulp build`

`gco .; gco 5f657b2; npm i; bower i; gulp build`

### Actual behaviour
Error:
>events.js:160
      throw er; // Unhandled 'error' event
      ^
Error: scripts/vendor-7604658e.js: error: couldn't process source due to parse error
Unexpected token (104737:10)
    at transform (/home/viktor/fablab/smartcitizen-web/node_modules/gulp-ng-annotate/index.js:20:11)
    at Transform._transform (/home/viktor/fablab/smartcitizen-web/node_modules/gulp-ng-annotate/index.js:68:25)
    at Transform._read (/home/viktor/fablab/smartcitizen-web/node_modules/gulp-ng-annotate/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Transform._write (/home/viktor/fablab/smartcitizen-web/node_modules/gulp-ng-annotate/node_modules/readable-stream/lib/_stream_transform.js:172:12)

viktorsmari, 2018-05-16:
This is related to ngAnnotate in `gulp/build.js` erroring

dsmrs, 2018-07-23:
fixed no? @viktorsmari

viktorsmari, 2018-07-24:
Yes!

closed 2018-07-24 by dsmrs