fablabbcn/smartcitizen-web #318: Search bar issues

pral2a, 2018-05-31:
### Expected behaviour
After `enter` search should show loading bar and display the results
### Actual behaviour
Fails to show result. See video
https://www.dropbox.com/s/lrsv4sog70bkp7f/search_bug.mov?dl=0
### How to reproduce?
Scroll to the bottom of the Kit view, then type a search and click enter

pral2a, 2018-05-31:
Maybe not related but this just happened

[image]

dsmrs, 2018-06-01:
do you know how to reproduce it? on the develop branch I don't have any problem

pral2a, 2018-06-01:
To reproduce the actual issue (not the second comment)
1. Open this url http://staging.smartcitizen.me/kits/1438
2. Scroll down to the bottom fo the kit view AKA footer
3. Then go to the nav bar a write something on the search bar like _barcelona_
4. You will see how the search doesn't work

dsmrs, 2018-07-23:
fixed by #341

closed 2018-07-23 by dsmrs