fablabbcn/smartcitizen-web #378: Buttons re-styling

marcelirodriguez, 2019-04-24:
**Layout re-styling:**
[image]

**Padding proportions:**
[image]

**Buttons colors:**
- Grey: #eeeeee > Hover color: #DFDFDF

**Status alert color:**
- Red: #F43D4D

pral2a, 2019-05-21:
Somethings are missing, sorry for the lack of detail:

1. Match exactly the buttons style from @marcelirodriguez reference. Including icons and colors:

[image]

_Ask him for the required assets_

2. Bring the buttons to the `showKit` view (see red arrows)

[image]

pral2a, 2019-05-21:
If we prefer to bring a ready made icon library to the platform instead of @marcelirodriguez icons I am fine but we need to use it across the whole platform (except for the sensor specific icons)

viktorsmari, 2019-05-21:
Here is how to change the colors globally, using AngularJS Material:

https://material.angularjs.org/latest/Theming/01_introduction

https://material.angularjs.org/latest/demo/colors

Theme can be changed here:
https://github.com/fablabbcn/smartcitizen-web/blob/0edcb4b24cace7b716037ab91efec6a829f46073/src/app/app.module.js#L25

closed 2019-06-18 by tuengominh