fablabbcn/smartcitizen-web #384: Add kit setup selection modal

pral2a, 2019-05-14:
1. Add new Kit button should trigger modal on **2**

[image]

2. Selection button:

- The button for SCK 1.0/1.1 will go to the state the previous button was going on `/kits/new` 
- The button for SCK 2.0/2.1 will go to `start.smartcitizen.me` 

[image]

pral2a, 2019-05-20:
Change SCK 1.1 image

[image]

viktorsmari, 2019-05-20:
New img added in cf4fb4dd6a24c68c86b7d08177f161e8636df6ad

commit 1a314bb7 (2019-05-15): Select kit popup.

Missing new images.
Closes #384
closed 2019-05-20 by viktorsmari