fablabbcn/smartcitizen-web #389: Metrics icons in dropdown with wrong size

oscgonfer, 2019-05-29:
@viktorsmari 

### Expected behaviour

Normal sized icons on kit selection dropdown, in the time series view of the kit's profile

### Actual behaviour

Icons for metrics are oversized like in the image

[image]

### How to reproduce?

Go to https://smartcitizen.me/kits/5565, and select a metric in the dropdown to view. Happens in both Firefox and Chrome.

viktorsmari, 2019-05-29:
@tuengominh can this be related to the changes you did the other day?

tuengominh, 2019-05-29:
@viktorsmari I inspected already. From what I see this is not related to the changes I did (I changed the fixed size of class sensor_icon from previously 14px to 22px). These icons Oscar mentioned using another class select_image and <img/> tag. But let me fix this issue, I think they must use <md-icon> and the sensor_icon_selected class.

closed 2019-07-09 by tuengominh