fablabbcn/smartcitizen-web #407: Kits not showing on the map

oscgonfer, 2019-09-18:
### Expected behaviour

All kits should show when exploring the map in smartcitizen.me/kits

### Actual behaviour

The kits (I believe ID > 10000) don't show up when exploring the map when exploring the map if you only visit [the web](http://smartcitizen.me/kits).

[image]

If you visit a particular device (like [this one](https://smartcitizen.me/kits/10041)) then they appear:

[image]

viktorsmari, 2019-09-30:
@oscgonfer is this still an issue?

I think this is the same issue since the other day, some kind of caching?

Now I can see these devices.
I can also see devices 10.000 and 10.000, they are the ones in La Marina De Port
[image]

oscgonfer, 2019-09-30:
Just checked the latest kits and they show while browsing. I don't know how long they take to appear though, it might be something to check?

viktorsmari, 2019-10-09:
Here we can see for a new device, that one view has 1 more kits (20) vs (21)

[image]
[image]

viktorsmari, 2019-10-09:
Looking at the developer console, I can see the id of the kit on both pages:
[image]
[image]

But it is not represented on the map.
Notice also that the nr of the request is 1225 on one page, but 1226 on the other. Could be a clue.

pral2a, 2019-10-16:
Any news from this?

Can we discard this is an issue between `world_map` and `fresh_world_map`? In other words, this looks like a front-end issue only, right?

We are receiving around 2-3 user complains per week, that's why I'm writing a follow-up

viktorsmari, 2019-10-16:
I trying to narrow it down, and I think we can discard the `fresh_world_map` - It is not called on either `/kits/` or `/kits/:id`, so looks like we are not using it in this case.

The `/kits/` endpoint already has the missing kits info in the `data` object on map initialize.

https://github.com/fablabbcn/smartcitizen-web/blob/cdc9b3f31fbf08442bcfe6f73bd1fe61b2629cf9/src/app/components/map/map.controller.js#L162-L166

When I console log the `data`, I can see the missing kits, but they are not shown on the map when we visit `/kits/` only on `/kits/:id`

commit fd996884 (2019-10-17): Add 'new' tags for selectedFilters

Might fix #407
closed 2019-10-17 by viktorsmari