fablabbcn/smartcitizen-web #410: Weird block on homepage: block repeated?

Katearmstrong4, 2019-10-06:
Hi @viktorsmari @marcelirodriguez - this looks weird. Is it repeated or waiting new content?

[image]

viktorsmari, 2019-10-06:
@Katearmstrong4 What is weird about it?
@pral2a added this yesterday.

Katearmstrong4, 2019-10-06:
Content wise the photo isn't centred and the text is copied from the
section above.

-- 
*Kate Armstrong*
*Fab Lab Barcelona | IAAC*
Telegram/WhatsApp: +34675441150
Fab.City Global Initiative
Distributeddesign.eu <http://distributeddesign.eu/> Project Manager

See Legal Data protection and confidentiality
<http://iaac.net/legal-notice-data-protection-and-confidentiality/>
Ver Aviso legal de protección de datos y confidencialidad
<http://iaac.net/legal-notice-data-protection-and-confidentiality/>
Veure Avis legal de protecció de dades i confidencialitat
<http://iaac.net/legal-notice-data-protection-and-confidentiality/>

closed 2023-03-01 by oscgonfer