fablabbcn/smartcitizen-web #418: Visiting user page redirects to login

pral2a, 2020-04-29:
### Expected behaviour

When searching for a user and clicking on it I want to see the user profile even when I am not logged in

### Actual behaviour

When searching for a user and clicking on it I'm redirected to login

### How to reproduce?

[image]

viktorsmari, 2020-04-30:
Works fine for me (not logged in) on Linux, both Chromium and Firefox

tuengominh, 2020-04-30:
I also don't have the issue. I can see the user profile when I am not logged in (Chrome and Safari).

pral2a, 2020-04-30:
I found the issue. It looks like it's linked to the auth cookie persisting in your browser even after you logged out. Just deleted all the site cookies and the issue vanished away.

So it might be linked with the log out function.

https://github.com/fablabbcn/smartcitizen-web/blob/ffc651576d5ef0dfd1c1247bde2dac5ec5df64a4/src/app/core/api/auth.service.js#L133-L135

Could you try to replicate the issue?

pral2a, 2020-04-30:
Yes, log in, log out and then do a search to visit a user profile. It it fails then clear the cookies and try it again.

tuengominh, 2020-05-14:
Hi @pral2a 

Sorry for getting back late to this. I tried to replicate the issue by logging in, logging out, and then searching to visit a user profile, but it worked fine for me. Do you still have the same issue?

Thanks

closed 2025-05-21 by oscgonfer