fablabbcn/smartcitizen-web #422: Users cannot delete a device - REMOVE button not working

viktorsmari, 2020-11-26:
### Expected behaviour
Device should be deleted

### Actual behaviour
Button does nothing - errors in dev console

### How to reproduce?
1. Log in
2. Go to your profile https://smartcitizen.me/profile/kits
3. Try removing a device by clicking the **REMOVE** button

[image]

Error:
```
TypeError: $mdDialog.confirm(...).title(...).content is not a function
    at Object.removeKit [as remove] (app-27d3da4d1b.js:5988)
    at fn (eval at compile (vendor-fec684fdfd.js:25632), <anonymous>:4:286)
    at callback (vendor-fec684fdfd.js:38165)
    at Scope.$eval (vendor-fec684fdfd.js:28607)
    at Scope.$apply (vendor-fec684fdfd.js:28706)
    at HTMLButtonElement.<anonymous> (vendor-fec684fdfd.js:38169)
    at HTMLButtonElement.dispatch (vendor-fec684fdfd.js:4435)
    at HTMLButtonElement.elemData.handle (vendor-fec684fdfd.js:4121)
    at HTMLButtonElement.d (raven.js:522) undefined
```

viktorsmari, 2020-11-26:
Problem solved in #58dbf03c by removing the `content` function

oscgonfer, 2021-01-18:
Hi @viktorsmari,

Trying this again in chrome, it doesn't seem to delete it either. Testing with device `13368` gives the following console error:

[image]

viktorsmari, 2021-01-18:
Where are you deleting the kits? Which URL?

If I delete them here, it works with no console errors:
https://smartcitizen.me/profile/kits

oscgonfer, 2021-01-18:
Hi, 

Sorry for the lack of detail, it's in the device page itself, in this case would be scrolling down, right below the graph (this would be another case where the button is):

[image]

viktorsmari, 2021-01-18:
Turns out the method for removing kits lives in 2 places. I had only fixed one of them before.

- `showkit.controller`
- `myprofile.controller`

Ideally I would refactor this into having only 1 method (DRY principle) but I don't know how to do it in AngularJS.
There is probably a reason for it living in 2 places, maybe we will find out later.

Method now updated in both places :+1:

oscgonfer, 2021-01-18:
Just tested it and works, great! We can close it for now and reopen when you want to refactor?

closed 2021-01-18 by oscgonfer