fablabbcn/smartcitizen-web #425: Include hardware ID in sensor profile

oscgonfer, 2021-01-23:
For the new workflows regarding data processing, we need to have a connection between the hardware itself and the devices on the platform. Some of the IDs in the hardware can't be accessed from the Data Board, and it's dangerous to have them written in it as the board itself can be replaced. Our way to connect them is through a hardware ID that we can manually put on the device and afterwards put it in `postprocessing_info` in the hardware field:

```
"postprocessing_info":
  {
   "updated_at": "2021-01-16T17:49:24",
   "hardware_url": "https://raw.githubusercontent.com/fablabbcn/smartcitizen-data/master/hardware/SCAS210010.json",
   "blueprint_url": "https://raw.githubusercontent.com/fablabbcn/smartcitizen-data/master/blueprints/sc_21_station_module.json",
   "latest_postprocessing": "2021-01-16T17:45:00"
  }
```

Currently, this can be updated only from two different places:

1. Via Curl manually, patching the postprocessing info of a certain device
2. Via `scdata` after processing the data of the device

This issue starts the conversation to implement a third option in which the user can add this manually through the kit edit view. The proposal would be to have a field in which the user pastes a valid json containing the `blueprint_url` field and the `hardware_url` field:

```
  {
   "hardware_url": "https://raw.githubusercontent.com/fablabbcn/smartcitizen-data/master/hardware/SCAS210010.json",
   "blueprint_url": "https://raw.githubusercontent.com/fablabbcn/smartcitizen-data/master/blueprints/sc_21_station_module.json"
  }
```

This field would need to:

1. Verify the JSON is valid
2. Potentially verify that the urls behind each field are accessible and also a valid JSON

An idea of how this would look like would be this:

[image]

pral2a, 2021-01-25:
Minor detail: like in Kit tags, field description supports links, so we can add the full documentation in [docs.smartcitizen.me](https://docs.smartcitizen.me) and link it here

oscgonfer, 2021-01-28:
Hey @viktorsmari, just checking if you noticed this.

viktorsmari, 2021-01-28:
> Hey @viktorsmari, just checking if you noticed this.

Noticed! Simply a matter of prioritization if / when I should work on this :nerd_face:

pral2a, 2021-01-28:
I think it's quite important as we're currently shipping quite a lot of Station that could benefit from it

oscgonfer, 2021-02-03:
Hey, 

Just reopening this to check if we can add links to the documentation and change the description in the new section:

https://github.com/fablabbcn/smartcitizen-web/blob/7529a70368aa6de12dffb5082b51e18ad32a63fb/src/app/components/kit/editKit/editKit.html#L174

Something like this would be better:

> Follow the instructions [here](https://docs.smartcitizen.me/Guides/data/Handling%20calibration%20data/) to generate a valid JSON containing the postprocessing information for your device. This is an advanced feature and it's not required for standard Smart Citizen Kits!

oscgonfer, 2021-02-03:
Done via https://github.com/fablabbcn/smartcitizen-web/commit/dae2d85a669b226be752fdac6cfcf93fdd182fcc

closed 2021-02-03 by oscgonfer

oscgonfer, 2021-03-25:
Regarding this, postprocessing has now a single point of entry (hardware ID), so we can delete the blueprint field and just leave the hardware info field. Only question left is how we are going to name it. Options:
- Hardware ID
- Hardware URI
- Hardware descriptor
As it stands right now, this field can contain an URL pointing to a json file, or a simple ID that can be found in [this folder](https://github.com/fablabbcn/smartcitizen-data/tree/master/hardware). This will surely change in the future if we scale it, but as it stands right now, this helps for flexibility of the postprocessing. 

Any comments?

With @pral2a we'll also work on adding this in the `Advanced Kit Selection` view of the onboarding for faster use. I'll document everything when it's all done.

commit 7529a703 (2021-01-29): Allow adding Postprocessing Info in kits/edit

Closes #425