fablabbcn/smartcitizen-web #427: Update data policy

oscgonfer, 2021-03-09:
@pral2a @viktorsmari 

Reviewing [policy](smartcitizen.me/policy), I found that there are some links that do not resemble the current status of the project. Some customers have asked about this policy, and would be good to update these links.

Concretely:
- Terms of use (easy fix)
- Purchase kit (with analytics) opens a popup window with old information

Questions:
- Do you want to keep analytics still?
- Do we want a pop-up?
- Do you want to update the pop-up just customize that with a text as such?
`The Smart Citizen Kit is currently commercially available through SEEED Studio (link) for the Smart Citizen Kit 2.1 and the Starter Pack (link and link) and directly via Fablab Barcelona (mailto) for more customised solutions such as the Smart Citizen Station and others`

viktorsmari, 2021-03-09:
The first link is broken.

About analytics, I have recently been looking into **stop** using Google Analytics altogether. 
I see a lot of people doing this.

In many of our websites we have GA installed and people rarely use it.
We could replace it with something simpler and more **privacy-friendly**

The main upside is we won't need a popup for our websites, and we are not allowing Google to spy on our users.

pral2a, 2021-03-09:
1. GA:

I'm support happy to remove GA. I actually want to suggest it. We don't need that amount of information. We can fully disable it even without an alternative. We can always get some insights about our traffic from the API backend metrics much more anonymised. 

2. Policy / Terms and Conditions

I will suggest we remove the existing policy and create a new fully based on the Make Works one

https://make.works/legal

Data controller should be IAAC as it is now.

2.1. Specific Data Terms

@oscgonfer it will be good to specify exactly the kind of Data License we use. We can add a section inside our Terms and Conditions.

https://help.data.world/hc/en-us/articles/115006114287-Common-license-types-for-datasets

closed 2023-03-01 by oscgonfer