fablabbcn/smartcitizen-web #443: Add pagination and search to user kits page

oscgonfer, 2023-07-20:
This is an issue to add a search bar and a pagination to the kits page for large amounts of kits. This could allow us to avoid slow page loads in the view with large amount of kits. Now, a quick and dirty profiling gives 1.2s of request time to the /me endpoint, but much more for a larger request:

[image]

oscgonfer, 2023-10-26:
We will move to the users endpoint for the requests on the users kit page. See referenced issue by @timcowlishaw 

@pral2a would you happen to know why we are asking for devices in this page?

pral2a, 2023-10-26:
I'm not sure if I'm getting the question fully. To fill the user profile with the devices owned by the user, we use /me to obtain the devices list and then /device/{id} to load all the devices a user owns. Originally, we expected a max of 10 devices per user. However, right now, some users own a lot of devices, and the view crashes, causing issues and becoming useless for those users.

oscgonfer, 2023-10-26:
> I'm not sure if I'm getting the question fully. To fill the user profile with the devices owned by the user, we use /me to obtain the devices list and then  /devices/{id} to load all the devices a user owns.

Yes, so we think is that we can query directly the `/users/{id}` endpoint to get all the information needed for the devices, as that endpoint provides already the information needed to populate the device-card. If we need something else, we can always change that api endpoint, but as I see it I think all is there.

pral2a, 2023-10-26:
Great! Please, ensure first how the end-point deals with private devices

oscgonfer, 2024-03-04:
Adding to this, we can improve the view by adding something like [`list.js`](https://listjs.com/).

closed 2024-03-19 by oscgonfer